mmakaay/dsmr-reader-for-esphome #5: RJ12 Partnumber

Sfunzer, 2022-12-28:
Hi!

An easy/quick one: do you have a part number for the RJ12 socket. I can't seem to find the right one.

Thanks!

thed6bel, 2022-12-30:
Hello, 
i ordered this one : 

https://fr.aliexpress.com/item/4000252589302.html?spm=a2g0o.order_list.order_list_main.165.3fa35e5b7ndciB&gatewayAdapt=glo2fra

:-)

PexJensen, 2023-03-18:
> Hello, i ordered this one :

And it fits perfectly.

Hoffnungsthal, 2023-11-23:
After a long search I found this one:

https://www.conrad.de/de/p/econ-connect-mjusn66gab-modular-einbaubuchse-mjusn66gab-buchse-einbau-horizontal-pole-6-schwarz-1-st-1311387.html

mmakaay, 2023-12-13:
I ordere mine off Ali. Two different sets. Black and grey ones. The black ones are the hardest to fit on the holes, but they do fit. They grey ones fit perfectly. So I guess as long as the footprint looks like the one on my board design, it should be okay.

closed 2023-12-13 by mmakaay